beegee-tokyo/DHTesp #1: Issue with library name

cmaglie, 2018-01-15:
Hi @beegee-tokyo 

we have detected that you changed the name of the library in `library.properties` from
  `DHT sensor library for ESP32`
to
  `DHT sensor library for ESPx`
is that right?
We are introducing a change in the library manager that forbids this kind of changes and ignore the releases with the name that differs from the name registered in the library index.

Is it ok for you to keep the new name `DHT sensor library for ESPx`? The older releases with a different name will be ignored by the library manager. In absence of an answer from you we will proceed with this proposal next week.

See also: https://github.com/arduino/Arduino/issues/7127

```
JOB 1239 - 2018/01/15 15:45:38 Scraping https://github.com/beegee-tokyo/DHTesp.git
JOB 1239 - 2018/01/15 15:45:39 Checking out tag: 1.0.3
JOB 1239 - 2018/01/15 15:45:39 Release DHT sensor library for ESP32:1.0.3 already loaded, skipping
JOB 1239 - 2018/01/15 15:45:39 Checking out tag: 1.0.4
JOB 1239 - 2018/01/15 15:45:39 Release DHT sensor library for ESPx:1.0.4 has wrong library name, should be DHT sensor library for ESP32
JOB 1239 - 2018/01/15 15:45:39 Checking out tag: 1.0.5
JOB 1239 - 2018/01/15 15:45:39 Release DHT sensor library for ESPx:1.0.5 has wrong library name, should be DHT sensor library for ESP32
```

beegee-tokyo, 2018-01-15:
Yes, please keep the new library name `DHT sensor library for ESPx`. I changed the name because I added compatibility with `ESP8266`, so `ESP32` in the library name wasn't correct anymore. I have no problems if older releases cannot be installed anymore.

Sorry for the problems I caused.

closed 2018-01-15 by beegee-tokyo

cmaglie, 2018-01-15:
No problems, thanks for the quick answer!