beegee-tokyo/DHTesp #2: ESP32Ticker.h conflict in ESP32 environment

9a4gl, 2018-01-22:
Hi,

great work.. you solved my issues with reading DHT22 on ESP32 platform. As soon I enables something wiht AsyncTCP (AsyncMqttClient or ESPAsyncWebServer), reading sensor failed.. with your aproach as in ESP32 example it works reliable.

Just using ESP32Ticker.h is a bit problem. It defines a class with name Ticker, so if you use standard arduino Ticker you get conflict.

BR,
Tihomir 9a4gl

9a4gl, 2018-01-22:
Actually you do not need that timer at all, it just complicates and it confused me. You can trigger reading each 2 sec from loop, and that is much simpler. IMO, examples should be simple.

9a4gl, 2018-01-22:
Not related to this issue. Have you tried https://github.com/pkourany/idDHT22Spark . It is interesting approach which also do not block main loop. Using interrupts should me more effective then creating task.

beegee-tokyo, 2018-01-23:
Thank you for the positive feedback

Arduino for ESP32 has currently no Ticker class and I plan to switch to the Arduino Ticker as soon as it is integrated. Which might happen soon -> [Ticker not implemented #30](https://github.com/espressif/arduino-esp32/issues/30#issuecomment-359493671)

I had a look on the Spark library and it still might fail if FreeRTOS decides to initiate a task switch and the new task locks interrupts. Same measure regarding task switch disabling will be necessary to make sure the reading from DHT is not interrupted.

ESP32 is a dual core uC with FreeRTOS as operating system. That's why I leave task switching to the OS, it (hopefully) knows better when to switch between tasks. Loop() is just another task, so not much difference if I read the sensor in Loop() or in a separate task. And in my application where I took the examples from, I have UDP broadcast listening, BLE server, TSL2561 light sensor, MQTT client and the DHT11 sensor and a TFT display running in parallel and I put all of these into separate tasks and interrupt or timer triggered. So far it is running for weeks now without any problems.

closed 2018-01-23 by beegee-tokyo

9a4gl, 2018-01-23:
I tried to find how often FreeRTOS switch tasks but could not, only how it behaves in case the task is suspended, blocked, etc.
Do you have any info or measurement of switching strategy ? In case tasks have same priority and fully running...

beegee-tokyo, 2018-01-23:
Check [RTOS Implementation](https://www.freertos.org/implementation/main.html) about the task switching strategy. There is a `RTOS tick interrupt` which is the period when RTOS checks if a task switching is necessary. However I do not know how long this period is or if it can be changed.    
When creating a task on ESP32 in Arduino framework, you can only give a priority, while in the FreeRTOS docs they talk about `When sleeping, an RTOS task will specify a time after which it requires 'waking'.`. I didn't find a possibility to set this time yet, but I am as well quite new to ESP32 and FreeRTOS.