beegee-tokyo/DHTesp #3: ESP32 example invalid rate

9a4gl, 2018-01-22:
In
https://github.com/beegee-tokyo/DHTesp/blob/master/examples/DHT_ESP32/DHT_ESP32.ino

    // Start update of environment data every 20 seconds
    tempTicker.attach(20, triggerGetTemp);

value is actually in microseconds, so it should be 20000000

beegee-tokyo, 2018-01-23:
Check ESp32Ticker.h you can see the function definitions:
```cpp
  void attach(float seconds, callback_t callback) {
    _attach_ms(seconds * 1000, true, reinterpret_cast<callback_with_arg_t>(callback), 0);
  }

  void attach_ms(uint32_t milliseconds, callback_t callback) {
    _attach_ms(milliseconds, true, reinterpret_cast<callback_with_arg_t>(callback), 0);
  }
```
attach() is seconds and attach_ms is milli seconds. There is no micro seconds ticker.

closed 2018-01-23 by beegee-tokyo

9a4gl, 2018-01-23:
Well, that is case if you are lucky and pick correct version of esp32 ticker. If you pick 
https://github.com/Pedroalbuquerque/Ticker-ESP32/blob/master/Ticker-ESP32.h
then you have microseconds :)
As there are many esp32 tickers, I suggest you to put link somewhere next to include to make sure people use correct one.

beegee-tokyo, 2018-01-23:
The reference to the library is actually in the beginning of the code:
```cpp
/**************************************************************/
/* Example how to read DHT sensors from an ESP32 using multi- */
/* tasking.                                                   */
/* This example depends on the ESP32Ticker library to wake up */
/* the task every 20 seconds                                  */
/* Please install Ticker-esp32 library first                  */
/* bertmelis/Ticker-esp32                                     */
/* https://github.com/bertmelis/Ticker-esp32                  */
/**************************************************************/
```
I hope the library will be included in the Arduino-ESP32 framework soon.