beegee-tokyo/DHTesp #4: Issue with DHT22 - Instant TIMEOUT and "nan" value

JavCaceres, 2018-02-03:
Hey! First, I wanted to thank you for the library that solved all my DHT and ESP32 compatibility problem. Huge work.

Anyways, I'm new using ESP32 and I have the following issue connecting it to a DHT22 sensor. The program uploads to ESP32 but it doesn't give me anything else than this. I'm using your DHT_Test.ino example for this. 
[image]

any ideas of what is happening? if it's something easy or stupid i beg for patience, i'm just learning from ESP32

JavCaceres, 2018-02-03:
Oh something else I forgot to say, is that the "nan" values keep coming out eternally, no real value is given during sensing, and I don't exactly understand either why does the TIMEOUT thing pop up...
Thanks for your time!

beegee-tokyo, 2018-02-03:
The automatic detection of the sensor type might not be working. Try to initialize the sensor with `dht.setup(2, DHTesp::DHT22);`
And do you have the pullup resistoron the data line? 4.7k or 10k is required.

JavCaceres, 2018-02-07:
Hello,
Sorry for the late answer; had to try a lot of stuff before writing a definitely answer. The problem was the automatic detection. For some reason I've never used DHT22 witouth resistors and it works perfectly in all the arduino's i've tried to the date. Just changing the code line you told me fixed the problem (without any resistor). Anyways, i'm going to use resistors know that I know they are necesssary for the cirtuit to avoid future problems. Thanks for your help!

beegee-tokyo, 2018-02-08:
Good! Thanks for the feedback. I will have to look into the automatic detection.

GanjaFarmer, 2018-02-28:
Hi, I use a Doit esp32 devkit v1 and dht22 with this nice library, auto detection worked for me. By the way on this board, pin 2 is used for the onboard LED, so it may be related to your problem.. Did you try another pin ?

beegee-tokyo, 2018-02-28:
Thanks for the positive feedback and the hint with the double use of the gpio. Very helpfull

closed 2018-02-28 by beegee-tokyo