beegee-tokyo/DHTesp #5: AUTO_DETECT doesnt work after a "cold" start

frankdownunder, 2018-03-02:
Power turned on .  .  . 
Status	Humidity (%)	Temperature (C)	(F)	HeatIndex (C)	(F)
OK	3.0		0.0		32.0		-3.9		25.0
OK	3.0		0.0		32.0		-3.9		25.0
OK	3.0		0.0		32.0		-3.9		25.0
Pressed tiny button on side of Wemos D1 mini , it starts to behave properly. . . .  . 
Status	Humidity (%)	Temperature (C)	(F)	HeatIndex (C)	(F)
OK	78.6		23.4		74.1		23.8		74.9
OK	78.8		23.4		74.1		23.9		74.9
I am convinced that AUTO_DETECT is implicated.  If I set the model explicitly, it seems to work better
`dht.setup(2, DHTesp::DHT22);`

beegee-tokyo, 2018-03-03:
I had actually a similar problem in my project [ESP32 Weatherstation](http://desire.giesecke.tk/index.php/2018/01/29/esp32-weatherstation/).
Not only the DHT11 but as well a TSL2561 gave wrong values after a cold start. I found out that when I delayed the initialization of the sensors and made it in loop() instead of setup() it worked all the time.
I guess, that after a cold start the FreeRTOS is eating up most of the CPU powers while setting up WiFi (and in my case BLE and Bluetooth). What I did is to put a delay in the start of the loop() to let things cool down, then initialize the sensors and let the app run.
```
bool sensorsEnabled = false;
void setup(void) {
   //.. do the main installation stuff like Serial, WiFi connection, Bluetooth, BLE, MQTT connection, ...
}

void loop() {
   // sensorsEnabled is false on first call of loop, then we wait 2 seconds, initialize sensor
   // After that sensorsEnabled is true and this is never called again until the next reboot or cold start
   if (!sensorsEnabled) {
      delay(2000); // Wait two seconds to let things settle down
      initTemperature();
      initLightSensor();
      sensorsEnabled = true;
   }
   // Do what the loop() has to do
}
```
Maybe not the most practical solution, but it works fine for me.

beegee-tokyo, 2018-03-11:
Autodetect might have not been worked because same issue as reported in https://github.com/beegee-tokyo/DHTesp/issues/6
I changed the example to use a different GPIO pin in release 1.0.6.

closed 2018-03-11 by beegee-tokyo