beegee-tokyo/DHTesp #6: Use of GPIO2 strapping pin in DHT_Test.ino example does not work on ESP32 M5Stack

krzychb, 2018-03-10:
Hi,

I could not load ESP32 M5Stack module using [default pin configuration](https://github.com/beegee-tokyo/DHTesp/blob/3eb3edb1ae0ce4104debc4cdcdad78dbf6ee55fd/examples/DHT_Test/DHT_Test.ino#L11) of DHT_Test.ino.

```
Archiving built core (caching) in: /tmp/arduino_cache_669293/core/core_espressif_esp32_m5stack-core-esp32_FlashMode_qio,FlashFreq_80,UploadSpeed_921600,DebugLevel_none_468c93602846210a69ed0642e5176d28.a
Sketch uses 165697 bytes (12%) of program storage space. Maximum is 1310720 bytes.
Global variables use 11136 bytes (3%) of dynamic memory, leaving 283776 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content
```

After changing to e.g. GPIO21, loading went without any issues and example works great!

I suggest changing GPIO2 to another pin, without strapping function, etc.

beegee-tokyo, 2018-03-11:
Thanks for the feedback. I will change the examples.

beegee-tokyo, 2018-03-11:
Updated example in last release to avoid this problem.

closed 2018-03-11 by beegee-tokyo