beegee-tokyo/DHTesp #7: Running two instances of DHTesp

rdheiliger, 2018-04-18:
I have two instances of DHTesp running on an ESP32, pins 26 and 27. Main loop calls below function each 5 seconds. Note that i had to put the delay(100) in the loop in order to get valid temp and humidity for the first instance. Have no idea why. Was getting 32 for Ambient and 1 for HumidityOutdoor, and believable numbers for the second instance.

void readHumidity(){

  HumidityOutdoor = outdoorDht.getHumidity();
  Ambient = outdoorDht.getTemperature();
  Ambient = outdoorDht.toFahrenheit(Ambient);

  delay(100);
 
  HumidityIndoor = indoorDht.getHumidity();
  House = indoorDht.getTemperature();
  House = indoorDht.toFahrenheit(House);

  Serial.print(outdoorDht.getStatusString());
  Serial.print("   Ambient   ");
  Serial.print(Ambient);
  Serial.print("   HumidityOutdoor   ");
  Serial.println(HumidityOutdoor);

  Serial.print(indoorDht.getStatusString());
  Serial.print("   House   ");
  Serial.print(House);
  Serial.print("   HumidityIndoor  ");
  Serial.println(HumidityIndoor);
}

beegee-tokyo, 2018-04-19:
To clarify, without the delay the readings of the outdoor sensor are wrong and the readings of indoor are correct?
Thats weird, I will look into it. I would have expected the problem the other way around. I dont have 2 sensors available, could you help me with testing if I give you some test code?

rdheiliger, 2018-04-19:
I just reloaded the program without the delay(100), and now both indoor and
outdoor are readint fine. Have not changed any other part of the program,
so, still don't know what caused it. Will look into a bit more tomorrow.
Thanks for the reply!

rdheiliger, 2018-04-19:
> Ok, let me know if you the problem comes back.
> Are you using DHT11 or DHT22? The DHT11 is really slow, after powerup the
> sensor should not be accessed for 1 second according to the data sheet. And
> response time for temperature change can be as long as 30 seconds, response
> time for humidity change can be as long as 15 seconds (according to the
> datasheet I have). So I am not sure if a reading every 5 seconds makes
> really sense.
>
> Don't know about the DHT22 limitations.

Using the DHT22. Problem is gone even on cold startup?? On site mentions
that reading faster than 2 seconds will overheat the device, so decided on
5 seconds, for my app, once a minute is likely enough. Thanks for looking!

beegee-tokyo, 2018-04-19:
Closing this as it seems to be solved. Feel free to reopen it if the problem is not gone.

closed 2018-04-19 by beegee-tokyo