beegee-tokyo/DHTesp #9: Missing esp_timer.h

horchi, 2018-06-18:
Hi,

one question about the dependencies of this library, at first the ESP32 Ticker was missing, I have installed it from here:  https://github.com/bertmelis/Ticker-esp32.git

Now, trying to compile the example/DHT_ESP32 get:
Arduino/libraries/Ticker-esp32/src/ESP32Ticker.h:27:24: fatal error: esp_timer.h: No such file or directory
Actually I have no idea where I can find the missing header. Any suggestions?

Best regards 
Jörg

beegee-tokyo, 2018-06-18:
Ups, thank you for the wake-up call. Ticker is now part of Arduino-ESP32 framework. Change ESP32Ticker.h to Ticker.h and it should work. Make sure you have the latest version of Arduino-ESP32 framework installed.

Need to update the example.

horchi, 2018-06-18:
i just installed the version 4.4.0 of the "core for ESP8266 WiFi chip"
And changed the header name, now I get:

Build-Optionen wurden verändert, alles wird neu kompiliert
dht22-test:22: error: 'TaskHandle_t' does not name a type
 TaskHandle_t tempTaskHandle = NULL;
 ^
/home/wendel/Arduino/dht22-test/dht22-test.ino: In function 'bool initTemp()':
dht22-test:53: error: 'tempTaskHandle' was not declared in this scope
       &tempTaskHandle,                /* Task handle. */
        ^
dht22-test:54: error: 'xTaskCreatePinnedToCore' was not declared in this scope
       1);                             /* Core where the task should run */

beegee-tokyo, 2018-06-18:
Are you testing for an ESP8266 or for an ESP32?

horchi, 2018-06-18:
ESP8266

beegee-tokyo, 2018-06-18:
Then `DHT_Test.ino `is the only example you can use.

horchi, 2018-06-18:
I'm pretty new to ESP8266 development, am I right in thinking that I have to use ESP8266 for these controllers and not ESP32? 
And, if I use the implementation according to DHT_Test.ino, dht.getTemperature () must be called in the frequency 'dht.getMinimumSamplingPeriod ()' or can it be called simply when the value is needed?

beegee-tokyo, 2018-06-18:
It can be called in whatever frequency you want. However the values returned will only be refreshed after the minimum sampling period which depends on the sensor used. On DHT11 that is 2 seconds e.g.

horchi, 2018-06-18:
Many thx for the explanation, helped me a lot to clarify!
And the kind how the get* methods are working is exactly what i'am looking for :)!

In meantime I had find out where my problem was - it depend on the auto detect of the sensor type, after 
 specify the type with dht.setup(io1, **DHTesp::DHT22**) it is working fine! before I got only 0 degrees Celsius and a humidity of 2 

Thx and best regards
Jörg

beegee-tokyo, 2018-06-19:
Updated the examples, the name now says for which board the example is. Added additional code that gives you a compiler warning if you try to compile an ESP32 example for an ESP8266.

Added as well a warning in the example (and in the DHTesp.h) that auto detect does not work (well).

closed 2018-06-19 by beegee-tokyo