beegee-tokyo/DHTesp #10: [FYI] Please refer more precise library.

nunobet, 2018-07-02:
https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTlib

This lib supports DHT33, DHT44 -very precise- sensors too.
Please refer more precise library.

beegee-tokyo, 2018-07-04:
I looked into the library, the way the DHT33 and DHT44 are accessed seems to be the same as for DHT22. But I have neither a DHT33 nor a DHT44, so I cannot test and confirm.
If you have one of these sensors, can you access it with **`dht.setup(dhtPin, DHTesp::DHT22);`** and tell me if it works. 
If it works, I can add DHT33 and DHT44 to the list of supported sensors.

nunobet, 2018-07-05:
Thank you for your reply.
Yes, dht.setup(dhtPin, DHTesp::DHT22); works well - whether the data is correct or not.
I think little lower value shows.

closed 2018-07-16 by beegee-tokyo