beegee-tokyo/DHTesp #11: DHT_ESP32 example with core 0 produce TIMEOUT error

aqucomp, 2018-07-02:
Hi,
if i change the core number to 0 in xTaskCreatePinnedToCore in your example DHT_ESP32, i take only TIMEOUT error.
I have change the sensor type to DHT22 too.
With original core 1 work perfectly.
Thx for your help.

beegee-tokyo, 2018-07-04:
Core 0 is doing all the WiFi stuff, FreeRTOS stuff like task switching and Bluetooth or BLE (if you activate it). I am trying to disable task switching while I read from the DHT sensor, but that might not work on Core 0.
Core 1 is running the Arduino loop() and is better suited to run such a fragile thing like bit-banging the DHT.

But your finding is good, I will add it to the README. Thank you.

And the auto detection does not work well, I marked it already as depreciated and will remove it sometime in the future.

aqucomp, 2018-07-04:
"I am trying to disable task switching while I read from the DHT sensor"
I know, that's why I'm using it :)

"but that might not work on Core 0"
What you mean "might"? It wont. Can you repair it? PLEASE.

I use ntp library too, another program. And if that just run dht.getTempAndHumidity in the midle from the ntp time select, its work!!??

"But your finding is good, I will add it to the README"
I hope, its not your last word.

Thx.

beegee-tokyo, 2018-07-06:
Actually, the example, if taken as here in the repo, will not work at all, because it has a big bug in it.
**`bool tasksEnabled = false`** is never set to true and therefor the sensor will never be read.

Thanks for guiding me to this mistake.

But I tried the corrected code using Core0 just now and it works fine. If you can share your code that fails when using Core0, I might be able to see what happens.

aqucomp, 2018-07-07:
hi,
"Actually, the example, if taken as here in the repo, will not work at all,"
hm.. it was/is working. 2 sec later. Just only on Core 1.
your old code is/was:
"void loop() {
  if (!tasksEnabled) {
    // Wait 2 seconds to let system settle down
    delay(2000);
    // Enable task that will read values from the DHT sensor
    tasksEnabled = true;  // !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
    if (tempTaskHandle != NULL) {
			vTaskResume(tempTaskHandle);
		}
  }
  yield();
}"

I just took your example.
But maybe it don't work within the loop() ? You just added tasksEnabled = true; in the setup(). Its very interesting.
I'm new to esp32, arduino, freeRTOS, and don't know all the dependency from freertos. Thats why I'm using library :) Don't you need to use semaphore or sth like that?

My own example was much simpler, without 2 places to reactivate the task. Without timer, just delay in while(1) loop from the task and delay(3000).
just in setup():
  dht.setup(dhtPin, DHTesp::DHT22);
  Serial.println("DHT22 initiated.");

  // Start task to get temperature
  xTaskCreatePinnedToCore(
      task_getTemperature,            /* Function to implement the task */
      "task_getTemperature",          /* Name of the task */
      4000,                           /* Stack size in words*/
      NULL,                           /* Task input parameter */
      5,                              /* Priority of the task */
      NULL,                           /* Task handle. */
      0);                             /* Core where the task should run 0 or 1 to tests*/
      
and the task had: while(1) loop
 with get_temperature
 serial.print
 and delay(3000)

beegee-tokyo, 2018-07-07:
delay(xx) is something that will always cause problems on a multi-tasking system. You are actually keeping your task alive, waiting for nothing, while other tasks might urgently need to be run.
That might work on Core1 (you are delaying just the loop() function. But on Core0 you have the FreeRTOS kernel, WiFi system task and eventually Bluetooth system tasks running. That's most likely the reason why your code crashes when run on Core0. You are blocking the system.

And that is why I use a timer in my example and send the temperature task into sleep as soon as I have read from the sensor.

And you are right, if you want to use the data outside of the temperature task, (in loop() or in another task), then you need to implement semaphores to protect the data from being accessed/changed by two tasks "simultaneously".

I kept the example simple, because it is not about how to write thread safe code, but to show how to get the data from the sensor.

beegee-tokyo, 2018-07-07:
And a small tip which makes reading the issues much easier. There is a "code" button above
[image]
If you past code just as you did, it is difficult to read. But if you paste your code, select it and then push this small button it is much easier to read
```
dht.setup(dhtPin, DHTesp::DHT22);
Serial.println("DHT22 initiated.");

// Start task to get temperature
xTaskCreatePinnedToCore(
task_getTemperature, /* Function to implement the task /
"task_getTemperature", / Name of the task /
4000, / Stack size in words*/
NULL, /* Task input parameter /
5, / Priority of the task /
NULL, / Task handle. /
0); / Core where the task should run 0 or 1 to tests*/
```

and the task had: while(1) loop
with get_temperature
```
serial.print
and delay(3000)
```

aqucomp, 2018-07-07:
"That's most likely the reason why your code crashes when run on Core0. You are blocking the system."
I don't think so. If it was true, the OS should crash, not the get_temp() procedure.

"delay(xx) is something that will always cause problems on a multi-tasking system."
I thought, that give the time back to the processor. Many examples in Network use it. hm..

But with your new example it still don't working too.

But i see now what i was changing and forget it. (I took new copy from github)
You wrote in your example, i should install this library: https://github.com/bertmelis/Ticker-esp32
The library is deprecated.
But I installed it and I have to change your program to compile, as an example in that Ticker library:
//#include "Ticker.h"
#include <ESP32Ticker.h>

Maybe that is the main problem?

"This Ticker library has just been merged into the Arduino core for ESP32"
And what is the right use of the library? And why you built your example on deprecated library?
Can't you please make an example on Arduino Standard Library? 

THX :)

aqucomp, 2018-07-07:
sorry, i didnt read your last post, before i answered to you. thx for the tip, next time i use it :)

beegee-tokyo, 2018-07-07:
Ok, I see. Firgot to remove that comment. Sorry for that.
At the time I wrote the lib, Ticker was not in the ESP32-Arduino framework. I changed the code 18 days ago, shortly after Ticker got available within the framework. Only then the TickerESP32 got deprecated.

beegee-tokyo, 2018-07-07:
When was the last time you updated your ArduinoIDE ESP32 installation? #include "Ticker.h" works without problem. Alternative you can try #include <Ticker.h>. And you can uninstall bertmelis library.

The latest version of the example is based on Standard Arduino without the need of additional libraries. It is only the comment that I forgot to change.

And I cltold you that your code most likely fails because of the delay(). When the system crash you see that it crashes inside get_temp(). That is the OS telling you where the problem occured.

aqucomp, 2018-07-07:
" because of the delay()" I've understand that, but in your own example you dont use delay(). 
So maybe its really the esp32ticker library. 
Ill be update my environment. Thx for that tip.
And i give you answer it it work now. Thx for help.

aqucomp, 2018-07-07:
hm.. i have make update for esp32 like in this doku:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md
but it not works. I still have compilation error: Ticker.h: No such file or directory

And in the Git Update I become:
[image]
Is that normal?

aqucomp, 2018-07-07:
after Merge->Merge local i take:
[image]

I've change the files in notepad++. I was tested the library with Serial.print().
Can you help me please.

beegee-tokyo, 2018-07-08:
You manually added stickbreakers I2C patch to your ESP32 installation. Thats why you cannot update.
stickbreakers I2C improvements are now merged into the ESP32 repo. 

You could now use [Add ESP32 to boards manager](https://desire.giesecke.tk/index.php/2018/07/01/add-esp32-to-boards-manager-workaround/) instead of manually install and update with Git

aqucomp, 2018-07-08:
How could you know, that was stickbreakers??? You are really good :)
And I change a little myself too :)

I've installed as the link says
[image]

but it doesn't work. I still have the error:
fatal error: Ticker.h: No such file or directory

What is wrong with my arduino?
How can i redo my changes to install from git gui? 
Or is there a button "force to overwrite" by update?

aqucomp, 2018-07-08:
I revert the changes:
[image]

so its look like:
[image]

I make manual again update and I have still the error with ticker.h not found :(

beegee-tokyo, 2018-07-08:
Not sure what is the problem with your ArduinoIDE. I just compiled the DHT_ESP32.ino example without problems. No error message regarding "Ticker.h"
[image]
Maybe you should cleanup the whole ESP32 installation of your ArduinoIDE.
1) delete what you installed by using Git
2) uninstall ESP32 with the board manager
3) close and restart ArduinoIDE
4) reinstall ESP32 with the board manager

aqucomp, 2018-07-09:
so, i manually deleted everything from Git in Arduino environment and try and is working. Its see, that git have upper priority. After i delete it i had to choose again the same bord.

Every test, that i made is working now, with delay() too :)

Thank you for your help. Very nice from you.

closed 2018-07-09 by aqucomp