beegee-tokyo/DHTesp #15: call of overloaded 'setup(int)' is ambiguous

GIOTLAB, 2018-08-07:
Hi,
I have been using this library for quite a while now and so far so good but now when I write:
`DHTesp sensor;
    sensor.setup(23);`
It's just not working for the ESP32.
I tried changing it to:
`DHTesp sensor;
    sensor.setup((uint8_t)23)`
But it's still not working.
It's outputting this error: call of overloaded 'setup(int)' is ambiguous/call of overloaded 'setup(uint8_t)' is ambiguous.

beegee-tokyo, 2018-08-08:
As auto detect of the sensor is not working well (sometimes ok, sometimes not, specially for the DHT11) I changed the setup() definition.
Try 
`sensor.setup(23, DHTesp::DHT11);`
or
`sensor.setup(23, DHTesp::DHT22);`
or
`sensor.setup(23, DHTesp::AM2302);`
or
`sensor.setup(23, DHTesp::RHT03);`
depending on the sensor you use.

GIOTLAB, 2018-08-08:
Thank you so much, it solved my problem.

closed 2018-08-08 by GIOTLAB

beegee-tokyo, 2018-08-08:
You are welcome.
And sorry, I should update the examples and the README. But I don't have the time right now.