beegee-tokyo/DHTesp #17: esp8266 example returning null for all fields

Octavian-the-ent, 2018-10-10:
I also can't get the functions [described here ](https://desire.giesecke.tk/index.php/2018/01/30/esp32-dht11/) to work correctly.

Octavian-the-ent, 2018-10-10:
Actually returning Timeout as the status, and then "nan" for every field

beegee-tokyo, 2018-10-10:
Sounds like a HW problem.
Did you try the sensor on an Arduino board or any other board to confirm it works?
Did you try another sensor?
Is the pullup resistor inside the sensor or external?
What value is the pullup sensor?
Did you measure all connections to make sure they are ok?
What board do you use? ESP32 or ESP8266?
What GPIO pin are you using?

Octavian-the-ent, 2018-10-11:
Ah, just a simple issue with my pin. My apologies!

beegee-tokyo, 2018-10-12:
I thought so.

closed 2018-10-12 by beegee-tokyo