beegee-tokyo/DHTesp #21: Issue with sub zero temp

vtomanov, 2019-01-10:
I have issue measuring sub zero temperature with DHT22 - on ESP32

the version of the lib is from :  11/MAR/2018

and when temp is sub zero is show zero nothing is minus scale ... ( Celsius ) 

any suggestions ?

beegee-tokyo, 2019-01-10:
No idea. To be honest, I never had my hands on a DHT22 nor the chance to test negative temperatures. The code for DHT22 is taken from other libraries.

I looked into the data sheet of the DHT22 and if temperature are negative then the highest bit of the transmitted data is set to 1.
The code takes this bit to convert the temperature into a negative value. 
```
    if ( rawTemperature & 0x8000 ) {
      rawTemperature = -(int16_t)(rawTemperature & 0x7FFF);
    }
    temperature = ((int16_t)rawTemperature) * 0.1;
  }

```
That should work, but I will give it a quick test tomorrow.

beegee-tokyo, 2019-01-20:
Sorry for late reply. 
I still have no DHT22 to test, but I used the code and injected values as `rawTemperature` that would give positive and negative temperatures. The code returned correct values.
Could you post some code how you read from the sensor? The functions to read temperature and humidity and the function to read both returns float values. Maybe a conversion error?

vtomanov, 2019-01-20:
I read it with :

DHTesp dht;
....
dht.setup(DHT_PIN, DHTesp::DHT22); // Connect DHT sensor to GPIO 17
 .....

float h =  dht.getHumidity();
float t = dht.getTemperature();

beegee-tokyo, 2019-01-21:
Looks OK.
No idea why it returns 0 on negative temperatures.
Seems I need to get my hands on a DHT22 to try. 

Could you try any other DHT library for Arduino?

vtomanov, 2019-01-21:
I have actually seen -0.4, -0.2 but not lower ....

beegee-tokyo, 2019-01-23:
Did you try another library. As I don't have a DHT22 (yet, ordered one now), I cannot guarantee function of DHT22.

vtomanov, 2019-01-26:
> Got finally a DHT22 (AM2302 actually) and did a test. As here in the
> Philippines it is difficult to find temperatures below 0'C I had to use my
> freezer.
> Test device: ESP8266
> Sensor: AM2302
> Test environment: Freezer compartment of my refrigerator.
> Data output: UDP broadcasting
>
> *I could measure temperatures down to -1.6'C.*
> It was difficult to have the serial port connected so I used a ESP8266
> powered by a power bank and transmitting the measured values over UDP
> broadcasting.
>
> I guess you have to check your sensor. The library works fine.
>
> Here is the result:
> DHT22.txt <https://github.com/beegee-tokyo/DHTesp/files/2798896/DHT22.txt>
>
> And here is the code snippet that I used:
>
> #include <Arduino.h>
> #include <ESP8266WiFi.h>
> #include <ESP8266WiFiMulti.h>
> #include <WiFiUdp.h>
> #include <DHTesp.h>
> [... 72 lines cut ...]
>       udp.endPacket();
>       udp.stop();
>     }
>   }
> }

-1.6 still OK ( I've seen -0.xx) but temperature was -10 when I measured it
and it was showing -0.xx but not anything closer to -10 - I guess it may be
the sensors but have try number of DHT22 and with small difference they
behave the same in - temp

e.g. it may be the DHT222 not your software

beegee-tokyo, 2019-01-26:
Best I could get with my freezer is -3.5 just now. But the library seems to work ok.

vtomanov, 2019-01-26:
Ok may be my sensors are bad - will order a new batch...

closed 2019-02-04 by beegee-tokyo

chank0, 2020-12-29:
I found a similar problem. I have been using the DHT22 with a mini D1 in a weather station, worked flawlessly for months. But, when a sub-zero temperature was reached for the first time, I started getting huge negative values (like -3200.0).

I did some research and found the following:

According to all the documentation I could find, the DHT22 sensor returns two bytes with the temperature. The first bit of the first byte is the sign ( 1 = negative, 0 = positive) and the rest of the bits are interpreted as an integer that is ten times the temperature. So, when we get a negative temperature, we just need to remove the first bit, read the remaining as an integer and then change the sign of the result.

However, I found that my sensor, when reading negative temperatures, returns all ones in the first byte. So, I get huge negative values:

Expected read: 10000000 01011100
Actual read:      11111111 01011100

(I got these readings by just putting the sensor in the freezer)

I tried to verify if the read from the sensor was correct (it appears to be, also no checksum problems). In the end, I worked it around by modifying my copy of the library so that if the sign bit is one, only the second byte of data is used to get the read. This limits the usable temperature range. Now, the sensor will not be able to read temperatures below -25.6 degrees (I dont expect it to read such temperatures, so it is not an issue for me)

The workaround I applied is here: https://github.com/chank0/DHTesp/commit/6e6d223577c4d5fe0756e3852848e5f991001527

I did not put a pull request because I understand the current implementation of the library is correct (it is doing what the sensor documentation says). I observed that my sensor does not behave as the documentation says, but I assume it is just a problem with my sensor. Anyway, I leave this comment here in case it is useful for anyone.