beegee-tokyo/DHTesp #24: using separate ADC

Pythonaire, 2019-03-07:
Hi,

because of the lousy ADC reading quality of my ESP (very noisy) I decided to use the MCP3008 ADC chip communication along the SPI interface. I tested that with a simple voltage divider and it works.
The MCP3008 library knows simple adc.Read("Channel"). 
Now, I like to use your library to connect an DHT22. How can I specify the right adc channel with your library? 
Thx

beegee-tokyo, 2019-03-07:
I don't understand your question. The DHT22 is sending the data as digital values, not as analog signal. There is no need to specify an ADC channel to it.

Pythonaire, 2019-03-07:
> I don't understand your question. The DHT22 is sending the data as digital values, not as analog signal. There is no need to specify an ADC channel to it.

Hahaha ... very sorry, yes that's true. reading the sensor documentation int a good idea. I would like to delete my question above, its very embarrassing for me.

beegee-tokyo, 2019-03-07:
You can edit and delete the content ;-)

closed 2019-03-07 by beegee-tokyo