beegee-tokyo/DHTesp #25: Instead of a pull-up resistor, consider pinMode(_pin, INPUT_PULLUP);

kissste, 2019-03-17:
Instead of a pull-up resistor, please consider:

pinMode(_pin, INPUT_PULLUP);

Works well for me.

P.S. This enables built-in pull-up resistors for GPIO 0-15. 
GPIO16 is down-only.

beegee-tokyo, 2019-03-19:
Might work on some uC but maybe not on all if the internal pull-up is too weak.
Would require to test with as many uC as possible and I don't have the time for it.

Maybe I can put it as an option when initializing the library.

beegee-tokyo, 2019-03-21:
I checked the internal GPIO pullup resistance for ESP32, ESP8266 and AVR uC.
The values are between 30k and 100k for ESP8266 and AVR or 10k to 100k for ESP32.

This might work for most applications, but if you have a uC at the upper resistance range (100k) I would say it will not work with long wires (DHT11/22 datasheet says up to 20m).

The DHT11 datasheet says 
**`When the connecting cable is shorter than 20 metres, a 5K pull-up resistor is recommended; when the connecting cable is longer than 20 metres, choose a appropriate pull-up resistor as needed.`**

closed 2019-03-21 by beegee-tokyo