beegee-tokyo/DHTesp #26: Reads nan all the time after deepSleep

cryptax, 2019-03-18:
Hi,
I have a **NodeMCU** v3 LoLin and a **DHT22** connected on **D4**. I am using your library to collect the values of temperature and humidity.

1. If I use your sample code (modifying pin to D4), just after upload, I get "**nan**" values from my DHT22 all the time **until I power it off and power it back in**. Then, I get nice values for temperature and humidity.
2. If I try to collect temperature and humidity, then go to **deepSleep**, **I always get "na" as return values** (even if I power on/power off). Note in that case for deepSleep to occur I cable RST to D0.
3. The **DHT22 and NodeMCU seem to be operating fine**. If I use the **micropython DHT library**, I get good return values straight away - no need to reboot/poweroff/poweron or whatever...

Any idea how to solve this?
Thanks!

This is the code I use for deepSleep:
```c
#include "DHTesp.h"

#ifdef ESP32
#pragma message(THIS EXAMPLE IS FOR ESP8266 ONLY!)
#error Select ESP8266 board.
#endif

DHTesp dht;

void setup()
{
  Serial.begin(115200);
  Serial.println();

  dht.setup(D4, DHTesp::DHT22); // my sensor is connected to D4
}

void loop()
{
  delay(dht.getMinimumSamplingPeriod());

  float humidity = dht.getHumidity();
  float temperature = dht.getTemperature();

  Serial.print(dht.getStatusString());
  Serial.print("\tHumidity %: ");
  Serial.print(humidity, 1);
  Serial.print("\t\tTemperature (C): ");
  Serial.println(temperature, 1);

  Serial.println(F("Entering deep sleep for 20 seconds"));
  ESP.deepSleep(20 * 1000000, WAKE_RF_DEFAULT); 
  delay(1000);  // wait for deep sleep to occur
}
```

beegee-tokyo, 2019-03-19:
I never used deepsleep function, so I am not 100% sure how it works.
If on wakeup it continues inside the loop() and is not calling setup again, the connection to the DHT might be broken and you would need to call dht.setup() after the delay(1000) call.

cryptax, 2019-03-19:
> I never used deepsleep function, so I am not 100% sure how it works.
> If on wakeup it continues inside the **`loop()`** and is not calling setup again, the connection to the DHT might be broken and you would need to call **`dht.setup()`** after the **`delay(1000)`** call.

Okay I'll try to re-do the dht.setup() in the loop and keep you posted. Thx.

beegee-tokyo, 2019-03-21:
Not sure what the problem is, but if I use D4 as you do, there is indeed a problem with reading the DHT sensor. And not only when using deepsleep() but even with the basic examples.

If I change to any other GPIO pin (I tried D2 and D5) it works, even with `dht.setup()` in the `void setup()` instead of the `void loop()`

Similar problem with D4 / GPIO2 was reported here
https://github.com/beegee-tokyo/DHTesp/issues/6
https://github.com/beegee-tokyo/DHTesp/issues/4

closed 2019-03-21 by beegee-tokyo

cryptax, 2019-03-26:
For your information, I switched to D5, and it works just perfectly fine. Something wrong with my D4 I guess...

beegee-tokyo, 2019-03-27:
Glad to hear its not the library.

D4/GPIO2 has special function during boot-up.

YorozuyaDev, 2020-03-01:
I had the same issue, I tried with differents pin in my nodemcu

beegee-tokyo, 2020-03-02:
I don't see any deep sleep call in your code.

Beside of that, your code should work if your sensor is working.