beegee-tokyo/DHTesp #34: Sensors without external pullup?

cr1st1p, 2019-09-01:
I'm trying to sue a DHT11, without an external pull up sensor, on an ESP8266.

Shouldn't the code always use `pinMode(pin, INPUT_PULLUP);`  instead of just 'INPUT' ?

beegee-tokyo, 2019-09-01:
On some devices (ESP32 e.g.) some GPIOs have no pullup function. That's why its not default. And on some devices the internal pullup resistor is not strong enough.

See issue #25

cr1st1p, 2019-09-01:
Thank you for pointing to #25 
So what I read is this:
a) you say that for most uC it might work, but if using dht with long wires (meters!) it will not work. And as such, it looks like you decided to not do anything and closed the report #25
Somehow I do expect most of the people are using dht sensors with wires waaay less than 20m.

b) You say here that esp32 have no pullup and that's why the default. Ok, probably. But esp8266 *does* have it. That is, *50%* of your library's scope (since you're saying this library is for esp8266 and esp32) !

c) well, and actually, you are not correct about esp32 and missing pull-ups. I'm not expert in it, but I think it does not have pullups on *some* of its pins, indeed (GPIO 34-39), but those can't be used as output, meaning they can not be used with dht anyway. 
  So... ??!?!

d) When you say 'the default' is to use 'INPUT' ... , it sounds to me that there should be also a non-default way of using it. But there is none implemented as far as I see

And I do expect that core Arduino  would do something smart about using INPUT_PULLUP even on a pin without a PULLUP.

beegee-tokyo, 2019-09-02:
You are welcome to submit a pull request with improvements to this freely distributed library. If you are not happy with what it provides and you are using ESP8266 you can as well try out one of the dozens of other DHT libraries, maybe another one suites your needs better. I optimized this library for the ESP32, because other DHT libraries are not working very well on this specific chip.

closed 2019-09-02 by beegee-tokyo