beegee-tokyo/DHTesp #35: Suggestion: reduce RAM usage by making ComfortProfile somehow optional

cr1st1p, 2019-09-01:
I'm not using the ComfortProfile related functionality, and as such, the whole structure is just taking 
precious RAM space inside the DHTEsp structure.
I wonder how people are using it? 

Maybe move it out of that structure and move the functions depending on it in the ComfortProfile.
Or have a DHTEspBase without any of the ComfortProfile things and DHTEsp derive from DHTEspBase + added ComfortProfile. 
Or make DHTEsp keep a *pointer* to such a structure? And maybe even allow use of PROGMEM stored structures. I doubt they're changing during execution time.

beegee-tokyo, 2019-09-01:
Not sure what "structure" you refer to. The comfort and perception functions use 2 enums with 17 bytes and 8 float values.
If you do not call the comfort and perception functions from your application, the compiler/linker will not include them into your final code.

Not sure how much bytes you will really save with your proposals.

For the PROGMEM on ESP32 that does nothing, not sure about the ESP8266.

If you feel comfortable, why don't you change the library according to your proposals, test it, and if it really saves alot, do a pull request and I will test it and merge it if it works.

cr1st1p, 2019-09-01:
I'm talking about member variable  DHTEsp::m_comfort.
Which has 8 floats. Enough bytes to not want to waste. And if you have more than one sensor, that data structure is not even shared and lost bytes multiplies.

Didn't know about esp32 using nicely const data from flash (haven't played with it yet) but that's only 50% of the scope of your library, 50% being esp8266 which *does* have the unfortunate PROGMEM feature.
And I did got into problems with not enough RAM with my program, so saving RAM when it easily can, that's good.
Compiler will cut *code* most of the time if not used, but I don't see it cutting the m_comfort structure.

beegee-tokyo, 2019-09-02:
You are welcome to submit a pull request with improvements to this freely distributed library. If you are not happy with what it provides and you are using ESP8266 you can as well try out one of the dozens of other DHT libraries, maybe another one suites your needs better. I optimized this library for the ESP32, because other DHT libraries are not working very well on this specific chip.

closed 2019-09-02 by beegee-tokyo