beegee-tokyo/DHTesp #36: Timeout problem when enabling wireless

rado99, 2019-09-10:
Hello,
I am facing very interesting problem. When I enable, no matter what (BLE, Wi-Fi or AP) wireless communication, I receive only the first measurement, the next ones are timeouts. When I comment this snippet, it works fine. I am using HiGrow board, so I am sure that the model of the sensor is DHT-11 and it is attached to pin 22.
Thank you in advance.

beegee-tokyo, 2019-09-10:
Difficult to say without seeing your code.
Are you running everything in loop() or are reading the sensor from a separate task? Please give some more details.

rado99, 2019-09-10:
I am running that switch in setup()
` switch(setting.mode){
    case BLE:setup_ble();break;
    case Wifi:setup_wifi();break;
    case AP:softAPinit();break;
  }`
Where mode is read from the Flash memory. I am sure, that I don't have problem, because the reading. When I comment this "switch" everything is fine. 
Snippet from setup_ble():
````
Serial.println("Starting BLE work!");
char buff[20];
  BLEDevice::init(setting.bleName);
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  BLEService *sensor = pServer->createService(SENSOR_UUID);`
....
  BLEDevice::startAdvertising();

```

beegee-tokyo, 2019-09-10:
Still no idea. I was using the DHT11 with WiFi and Bluetooth before.
Could it be that the GPIO for the DHT11 is assigned to different function in one of the init functions?

rado99, 2019-09-10:
For the Wi-Fi I am using AsynchronousHTTPClient 

> https://github.com/me-no-dev/ESPAsyncWebServer

, so it can be a problem, but I don't init it in BLE mode :/

beegee-tokyo, 2019-09-11:
I made a quick test with WiFi STA, WiFi AP, BLE with a custom characteristic and AsyncWebserver and it works without problems. There must be something in your code that I don't know about.
Here is the example that worked for me without problems:
```cpp
#include <Arduino.h>
// Includes for WiFi
#include <WiFi.h>
// Include for async webserver
#include <ESPAsyncWebServer.h>

// Includes for BLE
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLEDevice.h>
#include <BLEAdvertising.h>

// Includes for DHT
#include <DHTesp.h>

/** Pin number for DHT11 data pin */
int dhtPin = 22;
DHTesp dht;
TempAndHumidity newValues = {0, 0};

/** The server */
AsyncWebServer server(80);
/** Buffer for the generated server response */
char response[2048];

// List of Service and Characteristic UUIDs
#define SERVICE_UUID "0000aaaa-ead2-11e7-80c1-9a214cf093ae"
#define WIFI_UUID "00005555-ead2-11e7-80c1-9a214cf093ae"

/** Characteristic for digital output */
BLECharacteristic *pCharacteristicWiFi;
/** BLE Advertiser */
BLEAdvertising *pAdvertising;
/** BLE Service */
BLEService *pService;
/** BLE Server */
BLEServer *pServer;

/**
 * MyServerCallbacks
 * Callbacks for client connection and disconnection
 */
class MyServerCallbacks : public BLEServerCallbacks
{
	// TODO this doesn't take into account several clients being connected
	void onConnect(BLEServer *pServer)
	{
		Serial.println("BLE client connected");
	};

	void onDisconnect(BLEServer *pServer)
	{
		Serial.println("BLE client disconnected");
		pAdvertising->start();
	}
};

/**
 * MyCallbackHandler
 * Callbacks for BLE client read/write requests
 */
class MyCallbackHandler : public BLECharacteristicCallbacks
{
	void onWrite(BLECharacteristic *pCharacteristic)
	{
		std::string value = pCharacteristic->getValue();
		if (value.length() == 0)
		{
			return;
		}
		Serial.println("Received over BLE: " + String((char *)&value[0]));
	};

	void onRead(BLECharacteristic *pCharacteristic)
	{
		Serial.println("BLE onRead request");
		String wifiCredentials = " T:" + String(newValues.temperature) + " H:" + String(newValues.humidity);

		pCharacteristicWiFi->setValue((uint8_t *)&wifiCredentials[0], wifiCredentials.length());
	}
};

/**
 * initBLE
 * Initialize BLE service and characteristic
 * Start BLE server and service advertising
 */
void initBLE()
{
	// Initialize BLE and set output power
	BLEDevice::init("DHT-TEST");
	BLEDevice::setPower(ESP_PWR_LVL_P7);

	// Create BLE Server
	pServer = BLEDevice::createServer();

	// Set server callbacks
	pServer->setCallbacks(new MyServerCallbacks());

	// Create BLE Service
	pService = pServer->createService(BLEUUID(SERVICE_UUID), 20);

	// Create BLE Characteristic for WiFi settings
	pCharacteristicWiFi = pService->createCharacteristic(
		BLEUUID(WIFI_UUID),
		// WIFI_UUID,
		BLECharacteristic::PROPERTY_READ |
			BLECharacteristic::PROPERTY_WRITE);
	pCharacteristicWiFi->setCallbacks(new MyCallbackHandler());

	// Start the service
	pService->start();

	// Start advertising
	pAdvertising = pServer->getAdvertising();
	pAdvertising->start();
}

void initServer(void)
{
	Serial.println("Starting Async Webserver");

	//Async webserver

	// Show action page
	server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
		Serial.println("Received request to show top level");
		sprintf(response, "<h1>DHT test</h1><h2>Temperature %0.2f Humidity %0.2f</h2>", newValues.temperature, newValues.humidity);
		request->send(200, "text/html", response);
	});
	server.begin();
}

void setup()
{
	Serial.begin(115200);

	// Start WiFi STA
	WiFi.mode(WIFI_AP_STA);
	WiFi.begin("YOUR_AP", "YOUR_PW");
	uint32_t startTime = millis();
	while (WiFi.waitForConnectResult() != WL_CONNECTED)
	{
		if (millis() - startTime > 10000)
		{ // check if waiting time exceeded
			Serial.println("Cannot connect to WiFi");
		}
	}

	// Start WiFi AP
	WiFi.softAP("DHT11", "dht11_pw");

	// Initialize web server
	initServer();
	
	// Start BLE
	initBLE();

	// Initialize DHT sensor
	dht.setup(dhtPin, DHTesp::DHT11);
}

void loop()
{
	newValues = dht.getTempAndHumidity();
	// Check if any reads failed and exit early (to try again).
	if (dht.getStatus() != 0)
	{
		Serial.println("DHT11 error status: " + String(dht.getStatusString()));
	}
	else
	{
		Serial.println(" T:" + String(newValues.temperature) + " H:" + String(newValues.humidity));
	}

	delay(1000);
}```

closed 2019-09-25 by beegee-tokyo