beegee-tokyo/DHTesp #38: Reading values of NAN and TimeOut status of DHT sensor

Ninuss, 2019-10-05:
Hi, 
after the last update, my work sketches don't works anymore. 
I then used the example DHT ESP8266 sketch with both my ESP8266 and DHT22 (both working), but no longer reading the values. 
I only get (on serial monitor):

Status Humidity (%) Temperature (C) (F) HeatIndex (C) (F)
ESP8266_NODEMCU
TIMEOUT nan nan nan nan nan
TIMEOUT nan nan nan nan nan
TIMEOUT nan nan nan nan nan
TIMEOUT nan nan nan nan nan
...

  every 2000 ms (commented out the last cycle delay in the example sketch).

My sensor is connected to the GPIO0 pin of the ESP8266. (all wiring checked)

I tried another library and the sensor works and it reads the values reliably. 
I would like to use your library. 
Could you please check?
Thanks in advances.

beegee-tokyo, 2019-10-05:
I will check. There was an update recently based on a collaborator input. I didn't check it complete.
In ArduinoIDE you can choose to install an older version in the library manager. Can you go back one version and check if it works with it.

Ninuss, 2019-10-05:
Thank you again. I'll downgrade to a previus version until the next one will be published.

beegee-tokyo, 2019-10-06:
New version is out. I checked it on ESP8266 wit GPIO0 and it works fine again. Try version 1.17 of the library.

closed 2019-10-06 by beegee-tokyo