beegee-tokyo/DHTesp #39: Using this library using GPIO33 on ESP32

MorpheusDe97, 2019-11-28:
Hello, 

while playing around with this library, I accidentally wired my DHT22 to GPIO33 of an ESP-WROVER32-module. However, although the Pins GPIO32/33 seem to be connected to an internal XTAL, I still got results for temperatures and humidity, however I saw that there was a constant difference of about 2° plus a kind of small glitch that drives the temperature up over the time.

Can anyone explain this behaviour? Is it possible at all to get reliable values with GPIO Pins 32 or 33?

beegee-tokyo, 2019-11-28:
The data coming from the DHT22 is a digital value. It's validity is checked with a CRC checksum. There is no logical explanation why a different GPIO should deliver a different value.
The only reason I can think of is that the location of the sensor changed, maybe closer to a voltage regulator or power supply.

MorpheusDe97, 2019-11-30:
Okay, as I ran some tests today, comparing the temperature- and humidity values with a temperature logger, it just seemed to be a constant systematic error of 2.6°C, which I still find a bit odd, how I'll just count out the offset and leave it for now. 

Thank you for your quick answer @beegee-tokyo.

beegee-tokyo, 2019-11-30:
Thanks for the tests. Still find it odd that reading a digital value from different GPIOs shows differences in the temperature values.
Will close this issue, as I think it is not a problem with this library. 

Good luck with the DHT sensors, they are cheap (and not very reliable). For new projects I moved on to different types 8f sensors. Check SHT20 or BMP180/280 sensors. They are using I2C for communication, are not much more expensive and are more reliable.

closed 2019-11-30 by beegee-tokyo