beegee-tokyo/DHTesp #40: FreeRTOS doubts related with tempTask: vTaskSuspend(NULL) and xTaskResumeFromISR(tempTaskHandle)

vcmorini, 2019-12-25:
Hello :) 
Thanks for your examples!
Regarding the _DHT_Multi_ESP32.ino_ I have some questions, and I would be grateful if you could explain to me.
1) Why `vTaskSuspend(NULL)` is called? Is it the same as calling `vTaskSuspend(tempTaskHandle)`?

2) Why `xTaskResumeFromISR(tempTaskHandle)` is used instead of `xTaskResume(tempTaskHandle)` ?

Many thanks!

beegee-tokyo, 2019-12-26:
vTaskSuspend(NULL) suspends the running task, in this case the task reading the temperature.
VTaskSuspend(tempTask Handle) would be used e.g. from the loop{} {which is a task by itself) to suspend another task.

When I wrote the example I was not sure if the Ticker library is using interrupts therefore I used xTaskResumeFromISR.
As I know today xTaskResume would be working as well. 
xTaskResumeFromISR would be needed if the restart of the task would be triggered by an external interrupt, e.g. a button.

vcmorini, 2020-01-02:
Many Many thanks! 
Just another doubt concerning the same example.
Why core 1 is used? Could we also use Core 0?
Thanks a lot :)

beegee-tokyo, 2020-01-02:
Core 0 is running WiFi and BLE, so it might be to busy to handle the DHT sensor. But you can try.

vcmorini, 2020-01-04:
Got it, thanks!
For curiosity, how did you discover which core WiFi and BLE are using? I am using the FreeRTOS command: `xPortGetCoreID()`, but maybe you know another approach :)

Would you know if it is possible to change WiFi or BLE libs to Core 1, to balance things a bit more?

Many thanks for your input @beegee-tokyo

beegee-tokyo, 2020-01-04:
The info about core 0 is from some articles like this one https://esp32.com/viewtopic.php?t=9225

As far as I know,  in ESP32-Arduino FreeRTOS and the low-level WiFi and Bluetooth stuff is hard-coded to core 0.I do not know any method to change this. Maybe if you program in ESP-IDF you can change something with menuconfig, but I never tried it.

closed 2020-01-05 by vcmorini