beegee-tokyo/DHTesp #43: dht.getHumidity(); is causing a wdt reset on a ESP8266

vorkuta, 2020-11-12:
I'm using the example code for the ESP826, below.  

The line that is causing the issue is :

`float humidity = dht.getHumidity();`

The example code is:

`#include "DHTesp.h"

#ifdef ESP32
#pragma message(THIS EXAMPLE IS FOR ESP8266 ONLY!)
#error Select ESP8266 board.
#endif

DHTesp dht;

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println("Status\tHumidity (%)\tTemperature (C)\t(F)\tHeatIndex (C)\t(F)");
  String thisBoard= ARDUINO_BOARD;
  Serial.println(thisBoard);

  // Autodetect is not working reliable, don't use the following line
  // dht.setup(17);
  // use this instead: 
  dht.setup(7, DHTesp::DHT22); // Connect DHT sensor to GPIO 17
}

void loop()
{
  delay(dht.getMinimumSamplingPeriod());

  float humidity = dht.getHumidity();
  float temperature = dht.getTemperature();

  Serial.print(dht.getStatusString());
  Serial.print("\t");
  Serial.print(humidity, 1);
  Serial.print("\t\t");
  Serial.print(temperature, 1);
  Serial.print("\t\t");
  Serial.print(dht.toFahrenheit(temperature), 1);
  Serial.print("\t\t");
  Serial.print(dht.computeHeatIndex(temperature, humidity, false), 1);
  Serial.print("\t\t");
  Serial.println(dht.computeHeatIndex(dht.toFahrenheit(temperature), humidity, true), 1);
  delay(2000);
}
`

The output from the serial monitor is below:

`Status	Humidity (%)	Temperature (C)	(F)	HeatIndex (C)	(F)
ESP8266_GENERIC

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld`

vorkuta, 2020-11-12:
Note, am using an AM2302. Checked on [this site](https://desire.giesecke.tk/index.php/2018/01/30/esp32-dht11/) where the contributor to this project says that 'AM2302 is the same as DHT22'.

vorkuta, 2020-11-12:
Should also add - I am using the 3V pin on the ESP8266. I have tested with a meter and ~11.2 mV is flowing on the data line to pin 5.

beegee-tokyo, 2020-11-13:
The AM2302 is the same as DHT22.
None of what you wrote above should cause a WDT reset. 
Do you have a 10kOhm pullup on the data pin of the AM2302? 
The internal pull-ups of the ESP chips are not sufficient.

closed 2021-02-20 by beegee-tokyo