beegee-tokyo/DHTesp #48: AM2302 gives NAN

stevenveenma, 2021-04-06:
I have tried your library in the past with a DHT11 that worked quite well. But I am not able to succeed with a AM2302. The sensor is good, I tested it on an UNO with the Adafruit dht.h library. But I need an ESP 8266 as I need wifi. Of course I tested the right GPIO/PIN, I use a pullup resistor and I tried both using 3,3 and 5 Volt. In the sketch I tried dht.setup(14, DHTesp::DHT22) and dht.setup(14, DHTesp::AM2302). Moreover I tried version 1.18.0 and 1.17.0 of the library. I am running out of options now, any suggestions?

beegee-tokyo, 2021-04-07:
Sorry, no idea. To be honest, I completely abandoned the DHT11/22 sensors because there are much  better and nearly same price alternative sensors available.

foton, 2021-05-09:
I got the same problem `Temp: nan C;       Humidity: nan %`